opensensorhub/osh-core #1: Truncate ASCII values when significantDigits is specified

sensiasoft, 2015-10-17:
[image] **Issue by [mikebotts](https://github.com/mikebotts)**
_Friday Jan 09, 2015 at 00:25 GMT_
_Originally opened as https://github.com/sensiasoft/sensorhub-core/issues/1_

----

When SWE Common output components have the significantDigits property specified, truncate the output values according to the significantDigits value (e.g. 45.3049 versus 45.3049038746526).

alexrobin, 2017-02-23:
The plan is to do that with the new SWE data stream writer architecture that is also much faster

alexrobin, 2019-12-03:
Implemented in 12320c3

closed 2019-12-03 by alexrobin