opensensorhub/osh-core #8: [CLOSED] Add DoS protections

sensiasoft, 2015-10-17:
[image] **Issue by [sensiasoft](https://github.com/sensiasoft)**
_Saturday Feb 14, 2015 at 17:43 GMT_
_Originally opened as https://github.com/sensiasoft/sensorhub-core/issues/15_

----

It is currently very easy to overload the server by sending many data streams to it. Maybe some low level protections should be activated even when no security front end is used.

sensiasoft, 2015-10-17:
[image] **Comment by [sensiasoft](https://github.com/sensiasoft)**
_Saturday Feb 14, 2015 at 19:05 GMT_

----

First thing to do is enable options for activating Jetty DoS and QoS filters

alexrobin, 2019-10-31:
Added possibility to override default config with an external Jetty XML config file, which allows to setup the DOs filters.

closed 2019-10-31 by alexrobin