opensensorhub/osh-core #19: testGetObsTwoOfferingsByFoi fails on master

57Bravo, 2016-02-13:
I'm getting a failure of the unit test "TestSOSService::testGetObsTwoOfferingsByFoi" with my latest pull.

57Bravo, 2016-02-14:
Results :

Failed tests: 
  TestSOSClient.testConnectHttp:65 expected:<5> but was:<3>
  TestSOSService.testGetObsTwoOfferingsByFoi:526 Wrong number of observations returned expected:<2> but was:<6>

Tests run: 37, Failures: 2, Errors: 0, Skipped: 0

------------------------------------------------------------------------
Reactor Summary:

SWE Common Core ................................... SUCCESS [1:21.776s]
SWE Common OM and GML Models ...................... SUCCESS [28.199s]
SWE Common Image and Video Extensions ............. SUCCESS [5.521s]
SensorML Core ..................................... SUCCESS [33.948s]
OWS Framework Service Bindings .................... SUCCESS [1:13.059s]
SensorHub ......................................... SUCCESS [0.266s]
SensorHub Core .................................... SUCCESS [54.164s]
SensorHub PERST Storage ........................... SUCCESS [37.688s]
SensorHub SWE Services ............................ FAILURE [37.659s]
SensorHub Admin Tools ............................. SKIPPED
SensorHub Build ................................... SKIPPED
------------------------------------------------------------------------
BUILD FAILURE

sensiasoft, 2016-02-14:
Hi,
I cannot reproduce this problem with a fresh checkout of the master branch.
If you follow the steps in the developer's guide, osh-core should build without issues:

Clone a fresh version:
```
git clone --recursive https://github.com/opensensorhub/osh-core.git
```

Then do the maven build:
```
cd osh-core
mvn clean install
```

I did it in a brand new local maven repository just to be sure and I don't get any test failure.

On what platform are you trying to build?

57Bravo, 2016-02-14:
I walked through those steps at least five times blowing away the root each time to make sure I wasn't introducing something.
Same steps as you listed.
I also had a buddy of mine pull it and try to build it following the directions. He has the same issues that I do.
I'm building on windows 10. He's building on some flavor of linux.
He did say that he could get release 1 to build, but he had to fix a bunch of dependancies by hand.

https://github.com/opensensorhub/osh-core/issues/19#issuecomment-183904529

sensiasoft, 2016-02-14:
I'm still not able to reproduce your results but I see in your test results that your machine is running much slower than mine and I suspect you may be reaching some timeout values in some client-server tests. I have tried to remove all these timeouts so that test results are reproductible.

Could you please test building again with the latest osh-core from master?

57Bravo, 2016-02-16:
I removed my local copied, pulled (this evening), and built as listed above. 

I still get the following:

Results :

Failed tests:
  TestSOSService.testGetObsTwoOfferingsByFoi:528 Wrong number of observations returned expected:<2> but was:<7>

Tests run: 37, Failures: 1, Errors: 0, Skipped: 0

[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO]
[INFO] SWE Common Library ................................. SUCCESS [  9.372 s]
[INFO] SWE Common Core .................................... SUCCESS [01:24 min]
[INFO] SWE Common OM and GML Models ....................... SUCCESS [ 30.363 s]
[INFO] SWE Common Image and Video Extensions .............. SUCCESS [  5.204 s]
[INFO] SensorML Library ................................... SUCCESS [  0.078 s]
[INFO] SensorML Core ...................................... SUCCESS [ 34.935 s]
[INFO] OWS Framework Library .............................. SUCCESS [  0.109 s]
[INFO] OWS Framework Service Bindings ..................... SUCCESS [01:17 min]
[INFO] SensorHub .......................................... SUCCESS [  0.266 s]
[INFO] SensorHub Core ..................................... SUCCESS [01:00 min]
[INFO] SensorHub PERST Storage ............................ SUCCESS [ 54.314 s]
[INFO] SensorHub SWE Services ............................. FAILURE [ 29.533 s]
[INFO] SensorHub Admin Tools .............................. SKIPPED
[INFO] SensorHub Build .................................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 06:32 min
[INFO] Finished at: 2016-02-15T23:23:48-06:00
[INFO] Final Memory: 36M/127M
[INFO] ------------------------------------------------------------------------

mdhsl, 2016-02-26:
I cloned a fresh version and worked fine for me:

[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] SWE Common Library ................................ SUCCESS [7.731s]
[INFO] SWE Common Core ................................... SUCCESS [15.355s]
[INFO] SWE Common OM and GML Models ...................... SUCCESS [10.105s]
[INFO] SWE Common Image and Video Extensions ............. SUCCESS [1.323s]
[INFO] SensorML Library .................................. SUCCESS [0.013s]
[INFO] SensorML Core ..................................... SUCCESS [6.700s]
[INFO] OWS Framework Library ............................. SUCCESS [0.011s]
[INFO] OWS Framework Service Bindings .................... SUCCESS [9.930s]
[INFO] SensorHub ......................................... SUCCESS [0.042s]
[INFO] SensorHub Core .................................... SUCCESS [16.094s]
[INFO] SensorHub PERST Storage ........................... SUCCESS [3.400s]
[INFO] SensorHub SWE Services ............................ SUCCESS [12.474s]
[INFO] SensorHub Admin Tools ............................. SUCCESS [0.389s]
[INFO] SensorHub Build ................................... SUCCESS [16.073s]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 1:41.535s
[INFO] Finished at: Fri Feb 26 23:15:52 CET 2016
[INFO] Final Memory: 37M/244M
[INFO] ------------------------------------------------------------------------

mikebotts, 2016-06-02:
Its been a little while but did this ever get resolved for 57Bravo?  Thanks.