opensensorhub/osh-core #30: Automatically add a location obervable property to SOS offering

sensiasoft, 2016-06-02:
In order to simplify client interactions, we could automatically add an observed property for location/orientation every time a sensor advertises a feature of interest with location and/or its position (with location/orientation) in the SensorML description.

We could do this without creating additional sensor outputs or, alternatively, we can just use a status output for location/orientation when the sensor doesn't actually "measure" location.

mikebotts, 2016-06-02:
The goal here here I think is to provide consistency so the client always knows how to find the sensor location whether that sensor's location is static or dynamic, and whether its previously measured an documented or measured in real-time.

For that reason location/orientation should probably be provided as an advertised output.

Question 1: should orientation always be provided as well as location even if its undetermined and perhaps irrelevant (e.g. temperature detector)? Same question applies if location of the sensor is perhaps irrelevant or hidden (e.g. SIGINT, executed computer model, etc.).

Question 2: rather than sensor location, should the exposed location instead be observation location?  For in-situ sensors these are probably the same, but not the case for remote sensors and calculated or gridded observations.

alexrobin, 2016-07-22:
A location status output, an FOI and a position section in the SensorML document is now automatically created when location is provided in the sensor config (see example in FakeWeatherStation driver)

closed 2016-07-22 by alexrobin