opensensorhub/osh-core #36: Auto-purge function not reducing file-size of PERST storage

alexrobin, 2016-09-06:
When auto-purge is called on the PERST storage implementation, data is removed from the index but the file size is not reduced although objects are deallocated.

nicolas-f, 2018-06-12:
This issue is critical for me (on RPI)
Do you have alternative storage driver for org.sensorhub.impl.persistence.GenericStreamStorage ?

alexrobin, 2018-06-12:
Yes, there is an alternative embedded storage driver, based on embedded H2 database, that is both more reliable and faster: https://github.com/opensensorhub/osh-persistence/tree/master/sensorhub-storage-h2

We have been using it in production for several months now so it should be pretty stable. It hasn't made it to osh-core yet but it will replace the PERST implementation starting in version 2.0.

nicolas-f, 2018-06-12:
Ok nice.

I know H2, I've added Geometry type column.

There are some database corruption issues in MvStore so may be you should not set it as default storage for version 2.

I will try this driver thank you

alexrobin, 2018-06-12:
We are actually using MVStore directly, w/o the database layer at all. It has been working very well with very large datasets, including video. No corruption issues so far. Can you point me to these issues? What version is reported to have issues?

Regarding compilation, you'll have to compile it against the master branch of osh-core. We will release 1.3.2 soon.

nicolas-f, 2018-06-12:
Ah yes MvMap directly with spatial index, great

Ok, I will deploy snapshot version of osh on the prototype. Thanks

alexrobin, 2022-01-23:
This won't be fixed since the Perst DB driver has been deprecated in v2 in favor of the H2 driver

closed 2022-01-23 by alexrobin