opensensorhub/osh-core #41: Invalid KVP request. Please check your syntax

josephwinston, 2016-11-09:
I installed ```osh-v1.1.0-core-install.zip``` and then launched the program.  The following URLs work:

* <>
* <>
* <>
* <>
* <>
* <>

alexrobin, 2016-11-10:
Hello Joseph,

Everything looks normal. A request directly to the root of the SOS server is invalid. A valid KVP request (i.e. with arguments after the '?') has to be sent to it.

FYI, this HTTP endpoint implements [Sensor Observation Service](http://www.opengeospatial.org/standards/sos) standard interface published by the OGC (Open Geospatial Consortium). We wrote some documentation about it [here](http://docs.opensensorhub.org/user/sos/intro/).

Thanks for your feedback

closed 2016-11-10 by alexrobin

mikebotts, 2016-11-10:
Thanks.

---
Dr. Mike Botts
Botts Innovative Research, Inc.
http://www.botts-inc.com
mike.botts@botts-inc.com
256-652-0165

josephwinston, 2016-11-11:
Thank you for helping me with the issue.  

My confusion was the line ```You can connect to the SOS deployed at http://localhost:8181/sensorhub/sos right away, to get sensor data and metadata:``` from <http://docs.opensensorhub.org/install/>.

On second reading, this part of the sentence refers to the endpoints from the following list.  Maybe changing the introduction to ```You can connect right way to the SOS endpoints.  Here are some examples that work with the installed "fake" sensors.``` would help clarify the text.

alexrobin, 2016-11-11:
Ok, I see. Thanks for the suggestion. I updated the text of the install section.