opensensorhub/osh-core #47: Build failure: Execution failed for task ':swe-common-core:javadoc' under Unix

ch08532, 2016-12-02:
ENVIRONMENT: 

Buidling osh-core from a docker container (Ubuntu 14.04 LTS with Gradle 3.2.1 with Oracle JDK 8 latest)

STEPS:
The gradle build task fails on the following javasdoc task in swe-commong-core:

--
/opt/osh/repos/osh-core/lib-swe-common/swe-common-core/src/main/java/org/vast/unit/Unit.java:252: error: unmappable character for encoding ASCII
                                printSymbol = "???C";
                                               ^
/opt/osh/repos/osh-core/lib-swe-common/swe-common-core/src/main/java/org/vast/unit/Unit.java:252: error: unmappable character for encoding ASCII
                                printSymbol = "???C";
                                                ^
/opt/osh/repos/osh-core/lib-swe-common/swe-common-core/src/main/java/org/vast/unit/Unit.java:252: error: unmappable character for encoding ASCII
                                printSymbol = "???C";
                                                 ^
3 errors
:swe-common-core:javadoc FAILED

FAILURE: Build failed with an exception.
--

RESOLUTION: 

add JAVA_TOOL_OPTIONS=-Dfile.encoding=UTF-8 to your environment.

ex: 
export JAVA_TOOL_OPTIONS=-Dfile.encoding=UTF-8
gradle build

alexrobin, 2016-12-02:
I'm now setting the javadoc encoding option in the main gradle script. Can you let me know if that helps?

alexrobin, 2016-12-03:
Ok, I confirmed that the fix works when forcing the default system encoding to ASCII, so it should work for you as well

closed 2016-12-03 by alexrobin

ch08532, 2016-12-03:
Tested and works! Thanks!