opensensorhub/osh-core #49: Error when trying to getResults of an AXIS mjpeg -  Cannot instantiate class

ch08532, 2016-12-07:
I am able to connect to our AXIS cameras and see the PTZ data in the Admin UI (Note that I had to modify the axis driver to support VAPIX 3 API - the example axis driver is a VAPIX 2 implementation).  I am also able to get the results of the data using SOS getResult call however when I attempt to get the video using the same call I am getting the following error:

HTTP ERROR: 500

Problem accessing /sensorhub/sos. Reason:

    Cannot instantiate class

I can see in the logs that video data is flowing through the event bus:

2016-12-07 15:55:47.540 [EventBus-47] WARN  AsyncEventHandler - NEW_DATA_AVAILABLE Event from AxisVideoOutput to GenericStreamStorage @ 1481126147436, dispatch delay=104, queue size=4

Other than that I'm not seeing any errors in the logs.

ch08532, 2016-12-07:
UPDATE: I noticed that under SOS Service -> Custom Formats makes reference to two mime types to a class which do not exist in core but in osh-services repository (org.sensorhub.impl.service.sos.video.MJPEGSerializer).  After cloning the osh-services repo, building,then copying the services-video.jar file to my dist lib folder I am now able to get video results from the SOS (restart osh-core for changes to take effect).

alexrobin, 2016-12-07:
Yes, good catch. Things are a little scattered at the moment
I'm planning to regroup all the video stuff (drivers, helpers, services, dedicated storage implementation, etc.) in the same repo with better instructions.

alexrobin, 2017-01-25:
The SOS service now logs an error when custom formats handlers cannot be loaded

commit b02430dd (2017-01-25): Recreates proper configuration on sensor registration even if some parts
have been deleted. Also closes #22 and closes #49
closed 2017-01-25 by alexrobin