opensensorhub/osh-core #50: Add New Sensors Issue

CLEAOSBORNE, 2016-12-08:
When I add the sensor jar files to the core lib directory, the  new sensors will not populate or are able to be selected via the OpenSensorHub admin page.  Only the two sensors are included out of the box.....do you have to edit the config.json file to incorporate the new sensors or is there a work around?

alexrobin, 2016-12-12:
No, there is no configuration needed. This should work but you need to restart OSH after you drop the jars.

alexrobin, 2016-12-16:
Were you able to make this work?

CLEAOSBORNE, 2016-12-16:
Yes, thanks....I got it.  I did have an issue with building one specific distro though.  The osh-all distro failed to build and threw the error: Could not read script '/opt/osh-core/include-projects.gradle' as it does not exist.  This is the only distro that wouldn't build out.  Any ideas?  All the required osh-* dependencies are located in the same directory.

alexrobin, 2016-12-16:
Ok, I'm closing this issue then.
Regarding the osh-all distro, I haven't updated it after the latest changes on the build scripts so things won't build properly at this point...

closed 2016-12-16 by alexrobin