opensensorhub/osh-core #75: Admin UI displays poorly in chrome after adding a storage module

jamesyarrington, 2017-05-17:
Running OSH locally, using Chrome.

**To Reproduce:**
Create a new storage module, using Chrome:

1. Click "Storage" menu header.
2. Right click within storage menu.
3. Select "Add New Module".
4. Select "Real-Time Stream Storage", then click OK.

**Expected Result**
The just-created module should have its configuration displayed cleanly, as seen in Firefox:
[image]

**Actual Result**
The configuration window is not displayed properly in Chrome:
[image]

Also, after this bug occurs, the UI does not open new menus.  Clicking on "Sensors" will only close "Storage", not opening the "Sensors" menu.

This works fine in Firefox.

alexrobin, 2017-06-12:
I cannot reproduce this issue. What version of Chrome are you using?

jamesyarrington, 2017-06-14:
Version 58.0.3029.110 (64-bit)

jamesyarrington, 2017-06-14:
Just updated to:
Version 59.0.3071.86 (Official Build) (64-bit)

I'm still seeing the same behavior

alexrobin, 2017-10-10:
I still can't reproduce this bug. I'm now using Chrome Version 61.0.3163.100 (Official Build) (64-bit) on Linux Ubuntu 16.04. 

Can you check if it still happens for you in [v1.3.1](https://github.com/opensensorhub/osh-core/releases/tag/v1.3.1)?