opensensorhub/osh-core #116: Could not start SOS-T Client. Error while registering sensor with remote SOS with the pre-built demo android application

RohitMadhu, 2019-01-11:
Here is the error in logcat:

01-11 11:44:39.069 13146-15188/org.sensorhub.android E/o*.s*.i*.c*.s*.SOSTCli*: Could not start SOS-T Client
    org.sensorhub.api.client.ClientException: Error while registering sensor with remote SOS
        at org.sensorhub.impl.client.sost.SOSTClient.start(SOSTClient.java:244)
        at org.sensorhub.impl.client.sost.SOSTClient.handleEvent(SOSTClient.java:413)
        at org.sensorhub.impl.common.AsyncEventHandler$ListenerQueue$1.run(AsyncEventHandler.java:81)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588)
        at java.lang.Thread.run(Thread.java:818)
     Caused by: org.vast.ows.OWSException: Exception received from server
        at org.vast.ows.OWSUtils.sendPostRequest(OWSUtils.java:727)
        at org.vast.ows.OWSUtils.sendRequest(OWSUtils.java:616)
        at org.sensorhub.impl.client.sost.SOSTClient.registerSensor(SOSTClient.java:334)
        at org.sensorhub.impl.client.sost.SOSTClient.start(SOSTClient.java:239)
        at org.sensorhub.impl.client.sost.SOSTClient.handleEvent(SOSTClient.java:413) 
        at org.sensorhub.impl.common.AsyncEventHandler$ListenerQueue$1.run(AsyncEventHandler.java:81) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588) 
        at java.lang.Thread.run(Thread.java:818) 
     Caused by: org.vast.ogc.OGCException: Invalid Exception
        at org.vast.ogc.OGCExceptionReader.parseException(OGCExceptionReader.java:81)
        at org.vast.ows.OWSUtils.sendPostRequest(OWSUtils.java:721)
        at org.vast.ows.OWSUtils.sendRequest(OWSUtils.java:616) 
        at org.sensorhub.impl.client.sost.SOSTClient.registerSensor(SOSTClient.java:334) 
        at org.sensorhub.impl.client.sost.SOSTClient.start(SOSTClient.java:239) 
        at org.sensorhub.impl.client.sost.SOSTClient.handleEvent(SOSTClient.java:413) 
        at org.sensorhub.impl.common.AsyncEventHandler$ListenerQueue$1.run(AsyncEventHandler.java:81) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588) 
        at java.lang.Thread.run(Thread.java:818) 
     Caused by: org.vast.xml.DOMHelperException: Error while reading XML document
        at org.vast.xml.XMLDocument.parseDOM(XMLDocument.java:251)
        at org.vast.xml.XMLDocument.<init>(XMLDocument.java:113)
        at org.vast.xml.DOMHelper.parseStream(DOMHelper.java:1080)
        at org.vast.xml.DOMHelper.<init>(DOMHelper.java:129)
        at org.vast.xml.DOMHelper.<init>(DOMHelper.java:115)
        at org.vast.ogc.OGCExceptionReader.parseException(OGCExceptionReader.java:76)
        at org.vast.ows.OWSUtils.sendPostRequest(OWSUtils.java:721) 
        at org.vast.ows.OWSUtils.sendRequest(OWSUtils.java:616) 
        at org.sensorhub.impl.client.sost.SOSTClient.registerSensor(SOSTClient.java:334) 
        at org.sensorhub.impl.client.sost.SOSTClient.start(SOSTClient.java:239) 
        at org.sensorhub.impl.client.sost.SOSTClient.handleEvent(SOSTClient.java:413) 
        at org.sensorhub.impl.common.AsyncEventHandler$ListenerQueue$1.run(AsyncEventHandler.java:81) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588) 
        at java.lang.Thread.run(Thread.java:818) 
     Caused by: org.xml.sax.SAXParseException: expected: /hr read: body (position:END_TAG </body>@10:8 in java.io.InputStreamReader@afa1d6f) 
        at org.apache.harmony.xml.parsers.DocumentBuilderImpl.parse(DocumentBuilderImpl.java:146)
        at javax.xml.parsers.DocumentBuilder.parse(DocumentBuilder.java:107)
        at org.vast.xml.XMLDocument.parseDOM_JAXP(XMLDocument.java:361)
        at org.vast.xml.XMLDocument.parseDOM(XMLDocument.java:247)
        at org.vast.xml.XMLDocument.<init>(XMLDocument.java:113) 
        at org.vast.xml.DOMHelper.parseStream(DOMHelper.java:1080) 
        at org.vast.xml.DOMHelper.<init>(DOMHelper.java:129) 
        at org.vast.xml.DOMHelper.<init>(DOMHelper.java:115) 
        at org.vast.ogc.OGCExceptionReader.parseException(OGCExceptionReader.java:76) 
        at org.vast.ows.OWSUtils.sendPostRequest(OWSUtils.java:721) 
        at org.vast.ows.OWSUtils.sendRequest(OWSUtils.java:616) 
        at org.sensorhub.impl.client.sost.SOSTClient.registerSensor(SOSTClient.java:334) 
        at org.sensorhub.impl.client.sost.SOSTClient.start(SOSTClient.java:239) 
        at org.sensorhub.impl.client.sost.SOSTClient.handleEvent(SOSTClient.java:413) 
        at org.sensorhub.impl.common.AsyncEventHandler$ListenerQueue$1.run(AsyncEventHandler.java:81) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1113) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:588) 
        at java.lang.Thread.run(Thread.java:818)

alexrobin, 2019-01-11:
> at org.vast.xml.XMLDocument.(XMLDocument.java:113)
> at org.vast.xml.DOMHelper.(DOMHelper.java:129)
> at org.vast.xml.DOMHelper.(DOMHelper.java:115)
> (position:END_TAG @10 <https://github.com/10>:8 in

What URL are you using in the settings of the Android app?

RohitMadhu, 2019-01-11:
I am using the server ip address with the /sensorhub/sos endpoint

alexrobin, 2019-01-13:
Are you specifying the correct port as well? I believe it is set to 8181 in the default config.

It looks like the client is getting back an HTML error page instead of the XML response expected from server. Can you access the SOS capabilities page at `http://ip:port/sensor/sos?service=SOS&request=GetCapabilities`

RohitMadhu, 2019-01-15:
Hi Alex,

I am specifying the 8181 port. When I go to the page I get the 404 error:

HTTP ERROR 404
Problem accessing /sensor/sos. Reason:

Not Found

alexrobin, 2019-01-16:
With the default configuration, it should be `/sensorhub/sos`, not `/sensor/sos`
Which one are you trying to access?

RohitMadhu, 2019-01-16:
I tried the /sensorhub/sos one and this is what I am getting now.

`<sos:Capabilities xmlns:gml="http://www.opengis.net/gml/3.2" xmlns:ows="http://www.opengis.net/ows/1.1" xmlns:sos="http://www.opengis.net/sos/2.0" xmlns:swes="http://www.opengis.net/swes/2.0" xmlns:xlink="http://www.w3.org/1999/xlink" version="2.0.0">
<ows:ServiceIdentification>
<ows:Title>Test SOS Service</ows:Title>
<ows:Abstract>
An SOS service automatically deployed by OpenSensorHub
</ows:Abstract>
<ows:ServiceType>OGC:SOS</ows:ServiceType>
<ows:ServiceTypeVersion>2.0.0</ows:ServiceTypeVersion>
<ows:Profile>
http://www.opengis.net/spec/SOS/2.0/conf/resultRetrieval
</ows:Profile>
<ows:Profile>
http://www.opengis.net/spec/OMXML/2.0/conf/observation
</ows:Profile>
<ows:Profile>
http://www.opengis.net/spec/SOS/2.0/conf/sensorInsertion
</ows:Profile>
<ows:Profile>
http://www.opengis.net/spec/SOS/2.0/conf/sensorDeletion
</ows:Profile>
<ows:Profile>
http://www.opengis.net/spec/SOS/2.0/conf/obsInsertion
</ows:Profile>
<ows:Profile>
http://www.opengis.net/spec/SOS/2.0/conf/resultInsertion
</ows:Profile>
<ows:Fees>NONE</ows:Fees>
<ows:AccessConstraints>NONE</ows:AccessConstraints>
</ows:ServiceIdentification>
<ows:ServiceProvider>
<ows:ProviderName>Bobby Jones Inc.</ows:ProviderName>
<ows:ServiceContact>
<ows:IndividualName>Bobby Jones</ows:IndividualName>
<ows:PositionName>CEO</ows:PositionName>
<ows:ContactInfo>
<ows:Phone>
<ows:Voice>+1 800 123 4567</ows:Voice>
<ows:Facsimile>+1 800 123 4568</ows:Facsimile>
</ows:Phone>
<ows:Address>
<ows:DeliveryPoint>1 Main Street</ows:DeliveryPoint>
<ows:City>Portland</ows:City>
<ows:AdministrativeArea>OR</ows:AdministrativeArea>
<ows:PostalCode>97209</ows:PostalCode>
<ows:Country>France</ows:Country>
<ows:ElectronicMailAddress>admin@bobby.com</ows:ElectronicMailAddress>
</ows:Address>
</ows:ContactInfo>
</ows:ServiceContact>
</ows:ServiceProvider>
<ows:OperationsMetadata>
<ows:Operation name="GetCapabilities">
<ows:DCP>
<ows:HTTP>
<ows:Get xlink:href="http://localhost:8181/sensorhub/sos"/>
<ows:Post xlink:href="http://localhost:8181/sensorhub/sos"/>
</ows:HTTP>
</ows:DCP>
</ows:Operation>
<ows:Operation name="DescribeSensor">
<ows:DCP>
<ows:HTTP>
<ows:Get xlink:href="http://localhost:8181/sensorhub/sos"/>
<ows:Post xlink:href="http://localhost:8181/sensorhub/sos"/>
</ows:HTTP>
</ows:DCP>
</ows:Operation>
<ows:Operation name="GetFeatureOfInterest">
[... 217 lines cut ...]
<gml:upperCorner>34.8038 -86.7228</gml:upperCorner>
</gml:Envelope>
</sos:observedArea>
<sos:phenomenonTime>
<gml:TimePeriod gml:id="T2">
<gml:beginPosition>2019-01-03T19:07:32.380Z</gml:beginPosition>
<gml:endPosition indeterminatePosition="now"/>
</gml:TimePeriod>
</sos:phenomenonTime>
</sos:ObservationOffering>
</swes:offering>
<sos:responseFormat>http://www.opengis.net/om/2.0</sos:responseFormat>
<sos:responseFormat>http://www.opengis.net/om-json/2.0</sos:responseFormat>
<sos:observationType>
http://www.opengis.net/def/observationType/OGC-OM/2.0/OM_Observation
</sos:observationType>
<sos:observationType>
http://www.opengis.net/def/observationType/OGC-OM/2.0/SWEScalarObservation
</sos:observationType>
<sos:observationType>
http://www.opengis.net/def/observationType/OGC-OM/2.0/OM_ComplexObservation
</sos:observationType>
</sos:Contents>
</sos:contents>
</sos:Capabilities>`

RohitMadhu, 2019-04-08:
I finally figured out what was the issue, it seems to be note working though the corporate network, perhaps some sort of firewall. I just want to thank you for helping answer my questions, I really appreciate it!

closed 2019-04-08 by RohitMadhu