opensensorhub/osh-core #172: Memory leak with SOS websocket

alexrobin, 2021-06-14:
A memory leak becomes quickly visible when issuing many websocket requests. It impacts v1.4 and maybe v2.0

alexrobin, 2021-06-14:
It seems the issue is caused by a reference to the websocket session that is not nullified on close

mikebotts, 2021-06-14:
Alex. Does this affect how we should develop drivers or does it seem to be confined to core?

alexrobin, 2021-06-14:
No impact on drivers. It's confined to core (SOS impl). Mathieu is working on a fix.

mdhsl, 2021-06-17:
The memory leak appears when using the WebSocket service only. After investigation, we see that the listener of Jetty maintains a reference to the SOSWebSocketOut object which prevents it from being GC.

Next: https://stackoverflow.com/questions/55829525/potential-memory-leak-of-websocket-sessions-in-jetty-9-4-15

and updating Jetty to version 9.4.42.v20210604, the memory leak has gone.

The solution is simply to update Jetty

[image]

mikebotts, 2021-06-17:
Super. Good job. How might have this issue manifested itself in the server and in the COP?

closed 2021-06-21 by alexrobin