opensensorhub/osh-core #223: SensorWeb API - Some Systems not reporting due to validTime issue

drewbotts, 2022-07-15:
some systems do not show up in api/systems request but can be accessed by directly requesting api/systems/systemID#

alexrobin, 2022-10-04:
Request works when adding `validTime=../..`

ChainReaction31, 2022-10-04:
My issue is also related to this. Making the request with validTime=../.. works.

Server StackTrace:
```
2022-10-04 12:15:01.763 ERROR SWEApiService:2945f2ef [SWAPool-13] - Internal server error while processing request GET /sensorhub/api/systems (from ip=[0:0:0:0:0:0:0:1], user=admin)
java.lang.ArithmeticException: long overflow
	at java.base/java.lang.Math.multiplyExact(Math.java:949)
	at java.base/java.lang.Math.multiplyExact(Math.java:925)
	at java.base/java.time.Instant.toEpochMilli(Instant.java:1236)
	at org.sensorhub.impl.service.sweapi.feature.FeatureUtils.lambda$keepOnlyClosestToNow$2(FeatureUtils.java:83)
	at com.google.common.collect.Iterators$5.computeNext(Iterators.java:639)
	at com.google.common.collect.AbstractIterator.tryToComputeNext(AbstractIterator.java:140)
	at com.google.common.collect.AbstractIterator.hasNext(AbstractIterator.java:135)
	at java.base/java.util.Spliterators$IteratorSpliterator.tryAdvance(Spliterators.java:1811)
	at java.base/java.util.stream.StreamSpliterators$WrappingSpliterator.lambda$initPartialTraversalState$0(StreamSpliterators.java:294)
	at java.base/java.util.stream.StreamSpliterators$AbstractWrappingSpliterator.fillBuffer(StreamSpliterators.java:206)
	at java.base/java.util.stream.StreamSpliterators$AbstractWrappingSpliterator.doAdvance(StreamSpliterators.java:161)
	at java.base/java.util.stream.StreamSpliterators$WrappingSpliterator.tryAdvance(StreamSpliterators.java:300)
	at java.base/java.util.Spliterators$1Adapter.hasNext(Spliterators.java:681)
	at org.sensorhub.impl.service.sweapi.resource.BaseResourceHandler.list(BaseResourceHandler.java:270)
	at org.sensorhub.impl.service.sweapi.resource.BaseResourceHandler.doGet(BaseResourceHandler.java:91)
	at org.sensorhub.impl.service.sweapi.RootHandler.doGet(RootHandler.java:44)
	at org.sensorhub.impl.service.sweapi.SWEApiServlet.lambda$doGet$0(SWEApiServlet.java:148)
	at java.base/java.util.concurrent.CompletableFuture$AsyncRun.run(CompletableFuture.java:1736)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:840)
```

ChainReaction31, 2022-10-04:
Posting to : ```localhost:8282/sensorhub/api/systems```

with body: 
```
{
    "type": "Feature",
    "properties": {
        "name": "TEST",
        "description": "A new Test Sensor",
        "uid":"urn:osh:test:api",
        "type":"http://www.test.org/test/System",
        "definition": "http://www.test.org/test/System",
    }
}
```

creates the above issue when the GET request ```localhost:8282/sensorhub/api/systems``` worked prior.

alexrobin, 2022-10-06:
I'm not able to reproduce the issue.
@ChainReaction31 Can you try with the latest from v2 branch?

alexrobin, 2022-10-12:
@ChainReaction31 @drewbotts I just pushed a bunch of bug fixes to the v2 branch, including a fix that should prevent the overflow error Ian was seeing. Can you test with the latest version and let me know if it fully fixes your issues? Thanks

nickgaray, 2022-12-20:
Here is data from GeoHuntsville event exported from 1.3 and imported as 2.0

https://console.cloud.google.com/storage/browser/grx-datasets/GeoHuntsville_2017

commit 3dfea63c (2023-05-08): [SWAPI] Correctly list latest valid system description when listing
systems from multiple datastores (needed to compare full BigIds). Fixes
#223
closed 2023-05-08 by alexrobin