opensensorhub/osh-core #287: [CSAPI]/[CORE] Serialization of DataArrays includes incorrect "type" attribute in ElementCount

ChainReaction31, 2025-04-08:
When serializing data to be sent to the remote host, a DataArray with a variable size with encounter an issue where it has a `"type"` attribute. The remote API will accept a type but only if it follows an `"href"` attribute as that is how the Deserializer in the Binding determines a `Link` type versus a static `elementCount`. 

I created a simple fix that just swaps the order that those attributes are written, but the correct fix may be to not write the `type` at all in the case of elementCount or to allow it in the deserialization. OGC API Connected systems docs make it seem that a link is not expected to contain the type attribute so it probably should be dropped or ignored in some way.

ChainReaction31, 2025-04-08:
The fix I implemented as a work around.
```java
val = prop.getHref();
        if (val != null)
            writer.name("href").value(val);

        if (includeType) {
            writer.name("type").value("Link");
        }
```

alexrobin, 2025-04-09:
@ChainReaction31 did you try with the latest osh-core commit from master branch?
I think this should be fixed already.

ChainReaction31, 2025-04-10:
The branch with the client fixes is based off an older version of master than that fix then. I'll go ahead and close this.

closed 2025-04-10 by ChainReaction31