opensensorhub/osh-core #288: TLS 1.3 not completing handshake with TLS 1.2

drewbotts, 2025-04-09:
**Issue Description:**
With OSH nodes running Java 17 or 21 TLS has been updated to version 1.3. When trying to establish a TLS connection (using the TCP comm module with an OSH driver) to a server that is using Java 11 or earlier, the connection gets stuck in the handshake communication (this repeats indefinitely with no error message). 

**Attempts to Resolve:**
1. Downgrade TLS to 1.2 in launch script: I tried downgrading the TLS version as a flagin the launch.sh with `	-Djdk.tls.client.protocols="TLSv1.2" \
`  I monitored the communication in WireShark and it did seem to change the version to TLS 1.2 in the client side (OSH) TLS communications. The handshake continued to repeat without finalizing as before. 

2. Downgrade the TLS to 1.2 in the TCP comm module code: Downgraded the TLS version in the comm module with the following code `if (config.enableTLS) {
                    SSLContext sslContext = SSLContext.getInstance("TLSv1.2");
                    sslContext.init(null, null, null);
                    SSLSocketFactory factory = sslContext.getSocketFactory();
                    socket = factory.createSocket(addr, config.remotePort);
                    ((SSLSocket) socket).startHandshake();
                    is = socket.getInputStream();
                    os = socket.getOutputStream();`

This also did not change the repeated attempts to finalize the handshake.

**Additional Information Gathering:**
In order to confirm that the certificates were valid and that this was an issue based on the upgrade to Java 17 and 21 I ran an OSH node built with Java 11. The same driver was able to connect to the server without any trouble and ran as expected. I believe I was still using Java 21 on the launch.sh but the build of the node was done in 11.

My thought is that there is maybe still some differences in the Cipher Suites between the two communications and this may be causing the issue.

drewbotts, 2025-04-11:
I had tried that flag as well and it did not seem to have any affect on the handshake. 

I dont think we will have access or control to the server side of the connection to upgrade its TLS or Java (it is not an OSH node, but a different data server). I will continue attempts to set the right TLS version. My next step is to test re-enabling algorithms that were disabled on the Java 17 upgrade.

drewbotts, 2025-04-11:
I was able to get this specific issue resolved by editing my java.security file's disabled algorithms from:
`jdk.tls.disabledAlgorithms=SSLv3, TLSv1, TLSv1.1, DTLSv1.0, RC4, DES, \
    MD5withRSA, DH keySize < 1024, EC keySize < 224, 3DES_EDE_CBC, anon, NULL, \
    ECDH
to
`jdk.tls.disabledAlgorithms=MD5, SSLv3, DSA, RSA keySize < 2048, \
       rsa_pkcs1_sha1, secp224r1
       `

alexrobin, 2025-09-22:
Did you ever found the root cause for this?
Is this still an issue?